MangoAutomation/ma-core-public #1416: Modbus data source TCP Listener

Puckfist, 2019-04-22:
Modbus4J supports a TcpListener type of Modbus master, but it is not exposed through Mango. Some other scada softwares based on Mango M2M do support it, though.

It is for when Mango will still be acting as the master for protocol purposes, but where the device initiates the TCP connection.

minghu, 2019-11-28:
for  compromise  solution, the  System settings-->Virtual serial ports,can add one tcp Listener port Bridge to a virtual serial,so can  use it, and it suitable to all module,not only modbus,  of course,native support  is better。

minghu, 2019-11-28:
Some other scada softwares based on Mango M2M do support it, can you give a example,we can suggest it push the code.