MangoAutomation/ma-core-public #1417: is there necessary use the timestamp  to replace the longtime in pointvalues

minghu, 2019-04-24:
as all know,long time have the biggest time 2037 year problem,and mostly database native support timestamp type,is there necessary to  replace it?

terrypacker, 2019-04-24:
I assume you are talking about the 2038 year problem when the signed 32 bit integer value of seconds since Jan 1 1970 will overflow.

We don't use `TIMESTAMP` type columns which are 32 bits and will rollover in 2038.  All our timestamps are 64 bit integers, the `BIGINT` SQL type.

minghu, 2019-04-25:
thanks, I noticed that. [#1278](https://github.com/infiniteautomation/ma-core-public/issues/1278),that means in database stored  bigint type ,in api  levle use the date type, if use the unified type is better?

terrypacker, 2019-04-25:
We found that using 64 bit integers in our JSON models was cumbersome and rendering Date values as ISO Strings make more sense in most places.  

However if you are specifically talking about extracting Point Values from the API then the v2 endpoints have a parameter named dateTimeFormat:

closed 2019-04-29 by terrypacker