MangoAutomation/ma-core-public #1421: Double single quotes in i18n.properties

Puckfist, 2019-04-30:
If one is putting single quotes around a variable, one need use two single quotes. But, elsewhere, like in contractions, both single quotes will appear in the translation. For instance, send a test email from the email event handler and currently you will get 

This message was sent as a test of an event handler''s email recipient list

jazdw, 2019-05-02:
@Puckfist can you look into why this is? I'm pretty sure I don't need to double quote them for use on the front end but I have seen what you say on the backend.

Puckfist, 2019-05-15:
Ah, this appears to be caused particularly on those messages, which come from lines 154-17- of eventHandlers.jsp

As that is being deprecated, this issue can probably just be closed. My impression was perhaps that i18n syntax only required the double quote if it was around a variable like {0}, but I haven't read the specification on i18n syntax.

closed 2019-06-05 by Puckfist