MangoAutomation/ma-core-public #1423: TCP/IP data source retries don't really retry

Puckfist, 2019-05-02:
My testing was against a Linux machine using netcat.

TcpIpDataSourceRT

I found if I ctrl+C'ed a poll to close the connection, then the next time the data source attempted to poll the channel.write(command) call was draining the command by the buffer into the old channel. The channel did not realize it was closed until it attempts to read from the stream a few lines later, only to get an IOException. It then closes the connection and reopens it, but it does not have the command in the 'command' ByteBuffer any more! So, it does not resend the command.

Puckfist, 2019-05-15:
Found working this thread: https://forum.infiniteautomation.com/topic/4188/timeout-processing-with-tcp-ip-data-source

And additional recommendation involved checking for the termination character as the alleged "EOF" during reading.

terrypacker, 2019-07-01:
Released with Mango 3.6

closed 2019-07-01 by terrypacker