MangoAutomation/ma-core-public #1425: if  plan to official support postgresql?

minghu, 2019-05-08:
if  plan to official support postgresql?since it is not correct  running with postgresql.

terrypacker, 2019-05-08:
We do not officially support PostGresSQL, there are SQL database create files and a mechanism to load a PostGres driver but there has been no testing on it.  If you would like to add support for that it would be great!  Just fork and make a pull request once you have done some testing:

See these files for a start:
https://github.com/infiniteautomation/ma-core-public/tree/main/Core/db

If you can get us that far and confirm it works I will look at fixing up all the modules to support PostGres.

minghu, 2019-05-08:
ok,i can try it ,also, mango can consider provide commercial support for  postgres,bacause  the pointvalues  table is original suitable with  timescaledb(a time serials database base on postgres),then plus timescaledb  ecology.