MangoAutomation/ma-core-public #1426: DataPointRT savePointValueDirectToCache not so protected / consistent

Puckfist, 2019-05-08:
This method does not confirm that the value to be saved is of the proper data type, or not null, for instance.

Also, if the new value is the newest value then pointValue will not be updated. It is in the method comment that it does not fire events, but it would be odd to update pointValue without firing events, so perhaps there is a middle ground here?

terrypacker, 2019-05-09:
I've shadowed the method so that it remains the same and doesn't fire events.  Then I added another method with the desired sanity checks and a parameter that allows raising events in various scenarios.

```
    /**
     * This method is called by modules that have the potential to generate a rapid flow of values and backdates
     *  for the purpose of circumventing the update method's various controls on logging behaviors. It can generate events
     *  if desired.
     * 
     * @param newValue - the new value
     * @param source - for annotation
     * @param logValue - should the value be logged?
     * @param async - should this be done asynchronously i.e. queued in a batch 
     * @param fireEvents - how to fire events, 0=never, 1=if new value's ts is >= pointValue's ts, 2=always
     */
```

terrypacker, 2019-05-09:
Summary of locations that use this method for review:

1. AsciiFileDataSourceRT.fileEvents - This should fire events on update of value
2. HistoricalDataPoint.setPointValue (need to check references)
3. PointValueEmporter.importRow
4. DataFileDataSourceRT.updateDataPointValue
5. EgaugeDataSourceRT.queryHistory
6. HaystackHistoryImportResult.execute()
7. HistoricalMetaPointLocator.SetCallback.setImpl
8. DataImportController.importCsv
9. ScriptDataSourceRT.SetCallback.setImpl

terrypacker, 2019-07-01:
Closing as been released.

closed 2019-07-01 by terrypacker