MangoAutomation/ma-core-public #1438: AbstractDao MySQL issue

Puckfist, 2019-06-14:
I know a server that makes a lot of messages like this:

> ERROR 2019-06-14T11:22:12,662 (com.infiniteautomation.mango.db.query.StreamableSqlQuery.query:101) - You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ')  AND ue.userId=135  ) ORDER BY  activeTs  DESC LIMIT 50 OFFSET 0' at line 1 For Query: SELECT evt.id,evt.typeName,evt.subtypeName,evt.typeRef1,evt.typeRef2,evt.activeTs,evt.rtnApplicable,evt.rtnTs,evt.rtnCause,evt.alarmLevel,evt.message,evt.ackTs,evt.ackUserId,evt.alternateAckSource,u.username,(select count(1) from userComments where commentType=1 and typeKey=evt.id) as cnt ,ue.silenced FROM events  AS evt  LEFT JOIN  users u  ON evt.ackUserId = u.id LEFT JOIN  userEvents ue  ON evt.id=ue.eventId WHERE  ( evt.typeName=?  AND evt.typeRef1 IN ()  AND ue.userId=?  ) ORDER BY  activeTs  DESC LIMIT ? OFFSET ? 

Which is an syntax error in MySQL because "where x in ()" is not permitted (no in empty sets). The only place my initial investigation suggested was AbstractDao::applyConditions:588 as evt.typeRef1 does not appear in my workspace.

terrypacker, 2019-06-19:
@Puckfist that AbstractDao.applyConditions method is for the legacy UI and not related to `com.infiniteautomation.mango.db.query.StreamableSqlQuery`.  However I fixed a bug for this in 3.6 which may be what you are mentioning with this commit: 3e9ad26a29f4133b6fedf7067a3ad5aa7c7090c7

If you can give me the exact steps to reproduce this I can confirm it is fixed in 3.6.0.

Puckfist, 2019-06-19:
I did not produce it, a client who I believe is now on the latest Mango beta was producing the error output, which I do not see in their logs since the update, so you probably got it, and it probably wasn't where I suggested it could have been.

closed 2019-06-19 by terrypacker