MangoAutomation/ma-core-public #1439: User defined file locations

terrypacker, 2019-06-19:
Allow all application generated data and configuration file locations to be user defined.  Many locations used by Mango are based on MA_HOME, this is not ideal for being able to separate the application from its data.

Some things to review:

- [ ]  logfiles
- [ ] backups
- [ ] overrides
- [ ] module persist paths

jazdw, 2020-05-04:
We should get rid of the MA_HOME environment variable and start mango by pointing the process at the env.properties configuration file.

jazdw, 2020-08-20:
Another thing we should do as a part of this is to remove `overrides/properties` and just user `overrides/classes`

terrypacker, 2020-10-29:
I'm not sure that the FTL templates being at MA_HOME/ftl matters.  Those are packaged with the core and should not be modified.  The overridden files should be at `OVERRIDES/ftl`

terrypacker, 2020-10-29:
Ah nope never mind.  I'll leave that comment for reference but module FTLs are placed there on installation.

jazdw, 2020-12-04:
We should load the freemarker FTL files from the classpath, see this stack trace provided by @Pacane 
```
java.io.FileNotFoundException: /data/ftl does not exist.
	at freemarker.cache.FileTemplateLoader$1.run(FileTemplateLoader.java:122) ~[freemarker-2.3.28.jar:2.3.28]
	at freemarker.cache.FileTemplateLoader$1.run(FileTemplateLoader.java:119) ~[freemarker-2.3.28.jar:2.3.28]
	at java.security.AccessController.doPrivileged(Native Method) ~[?:?]
	at freemarker.cache.FileTemplateLoader.<init>(FileTemplateLoader.java:119) ~[freemarker-2.3.28.jar:2.3.28]
	at freemarker.cache.FileTemplateLoader.<init>(FileTemplateLoader.java:98) ~[freemarker-2.3.28.jar:2.3.28]
	at com.serotonin.m2m2.Lifecycle.freemarkerInitialize(Lifecycle.java:716) ~[ma-priv-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:285) ~[ma-priv-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.Main.main(Main.java:144) ~[ma-priv-4.0.0-SNAPSHOT.jar:?]
```

jazdw, 2020-12-09:
There's now a `paths.data` property which controls the variable data root, all databases, filestores etc are created under this directory. This defaults to `$MA_HOME` so the upgrade path is seamless.

jazdw, 2021-03-03:
Closing this, opened new issue for remaining tasks #1706

closed 2021-03-03 by jazdw