MangoAutomation/ma-core-public #1441: Import Point Values via CSV and REST

terrypacker, 2019-06-19:
This endpoint/UI will allow uploading a potentially large CSV/JSON/Excel? format file that is streamed into the database for one or many points. 

This document describes the existing formats that ideally we can use for this: https://help.infiniteautomation.com/point-value-v2-endpoints?rq=rest

The uploading user must have set permission for each data point that values will be set on.

Joelhaggar, 2019-06-19:
This should work with at least one of the watch list download formats.  I wonder if there is a way around the file size upload limit for this?

terrypacker, 2019-06-20:
I'm sure there are several ways to work around the max size where we could allow large uploads on this endpoint only.  The idea I'm thinking of is to override the `org.springframework.web.multipart.commons.CommonsMultipartResolver` which is what controls the size and let it allow larger sizes for a whitelist of endpoints...  Have to think about the security ramifications for this one.

terrypacker, 2019-06-20:
The more I think about it we will probably have to do something like that anyway as I'm not sure we will want to actually use a file upload with multi-part files, that will save a temporary file for us to process.  I was hoping to be able to catch the incoming request and stream its contents into our database without touching the disk first.  Much like how we can stream data out of the database.

This should do the trick: https://commons.apache.org/proper/commons-fileupload/streaming.html

terrypacker, 2019-08-29:
Finished https://github.com/infiniteautomation/ma-modules-public/pull/62

closed 2019-08-29 by terrypacker