MangoAutomation/ma-core-public #1443: Events Web Socket - Ability to query events

terrypacker, 2019-06-20:
There are two scenarios where it would be nice to get the number of active and/or un-acknowledged events via WebSocket
• For a specific data point when we are displaying its point value we display its active events, if we are on a Watchlist this currently results in 50 REST requests to get each point's events
• For displaying an un-acked events count in the toolbar, or for the event audio we have to first query for the event count then monitor the WebSocket

I think it could be 2 parts
• A welcome message that the server sends when the client connects that gives a detailed summary of the current event status, i.e. number of un-acked events of each level and maybe also number of active events of each level
• A way to retrieve/query for events relating to a data point / maybe a broad query functionality

The reason this is difficult currently is
a) I can't query the REST API using a single query as there are 50 components which all init at different times and all need an active event count
b) I need to synchronize the API call with any WS events which occur while making the query. i.e. if I start an API request and the WS in the meantime tells me that an event became active is this included in the API result or not? I have no idea.

terrypacker, 2019-06-20:
One design concept for the request/response models is:

Request, must supply both eventTypes AND levels or you get unsubscribed. 
sendEventLevelSummaries and query are optional.

```
{
  eventTypes: [ACKNOWLEDGED, RAISED],
  levels: ['URGENT','INFORMATION'],
  sendEventLevelSummaries: true,
  query: 'rql string'
}
```

Response, list of active summaries (or null if not requested) and the query results 
which are the same as a normal RQL query (or null if not requested).  This implies 
I will always send a response even if it is:
```
  {
    activeSummaries: null,
    queryResults: null 
  }
```

```
{
  activeSummaries: [ 
     {
        level: 'URGENT',
        unsilencedCount: 12,
        mostRecentUnsilenced: { EventInstanceModel }
     }, ...
     ],
     queryResults: {
        items: [EventInstanceModel,...]
        total: 100 
     }
}
```

terrypacker, 2019-07-13:
The work is being done in this branch: https://github.com/infiniteautomation/ma-modules-public/tree/events-rest-v2

Left to do:
[ ] - Add REST endpoints to the controller
[ ] - Add JOOQ support for all the various RQL query mappings that exist in the legacy controller

jazdw, 2019-07-15:
Why are you adding REST endpoints?

jazdw, 2019-07-15:
Also we should not send RQL queries as strings, we have a model for these. String RQL queries are for use in URLs.

jazdw, 2019-07-15:
Other things for discussion
- Why use a MultiSessionWebSocketHandler instead of PerConnectionWebSocketHandler?
- Listener termination should be wrapped in try/catch block to ensure super method is called
- It seems like I can create multiple event listeners per websocket session by sending multiple subscriptions

terrypacker, 2019-07-15:
Re: REST endpoints: no reason other than I had to make v2 models for everything so figured we could write v2 endpoints.  Happy to skip that part.

Re: RQL as Strings: fine by me wasn't sure how you wanted to transmit the query.  

Re: Why use a MultiSessionWebSocketHandler instead of PerConnectionWebSocketHandler? (like the old Events WS handler does): using all v2 apis.

Re: Listener termination should be wrapped in try/catch block to ensure super method is called: good call, will do

Re: It seems like I can create multiple event listeners per websocket session by sending multiple subscriptions: Will investigate that, agreed that should not be possible.

Re: Can we just implement this on the old WS? Why do we need a new one?
You specifically told me you wanted to use the new v2 web sockets for this with support for the new sequence/timing abilities.

jazdw, 2019-07-23:
- V2 models are needed as event type is already mapped to a new model in V2
- If we are going to use V2 models then we should have full rest controller support and move UI over to using V2 end points
- Probably remove RQL querying support and only support requests for getting:
1. Active events or active event counts for data points
2. Event level summaries
- RQL querying could return large amounts of data and block up WS
- Do **not** inject REST controller into WS, any logic which is common to WS and controller should be moved to a service and injected into both

terrypacker, 2019-07-23:
Summary of work left to do:

- [ ] Add REST endpoints to the controller
- [ ] Add JOOQ support for all the various RQL query mappings that exist in the legacy controller
- [ ] Move logic from REST controller into service in Core for use in web socket and controller
- [ ] Review model for Data Point Summaries (has counts per level for each point at the moment)

closed 2019-08-28 by terrypacker