MangoAutomation/ma-core-public #1455: Virtual serial ports  support mang serial use same port  identify by  register  code.

minghu, 2019-07-02:
when  serial server mode, can support mang serial use same port  identify by  register  code,and register code is also used as Security Code，this can refuse  Unrecognized  connect。
also,client mode or server mode,support ports status monitor，later, can bind a eventdetector  when one port disconnnected,can raise a alarm。

this is ok?

terrypacker, 2019-07-02:
@minghu it seems you are asking 2 things:

1.   For Virtual Serial Server Sockets you would like some sort of security added, can you elaborate perhaps with an example?

2.  You would like us to add events to the Virtual Serial Ports on connect and disconnect so that you can add event handlers for these types of events?  Is this correct?

minghu, 2019-07-04:
for 1:
       mang serial port user one same tcp server port can increase port  Utilization rate,also security .
 for example:
    com1:tcp server port:9000     register code(also used as  security code) :"this is com1，1111",
    com2: tcp server  port:9000    register code(also used as  security code) :"this is com2,2222",
    com3: tcp server  port:9000    register code(also used as  security code) :"this is com3,3333",

   when client connect to 9000,at once  send  the register code，so server can    identify  it，if one   Unrecognized connect  don't send the egister code,server can refuse it 。

for 2: it right, it useful to send alarm  when one Virtual Serial   disconnect