MangoAutomation/ma-core-public #1456: Rate limiter stops loading of user module files/logos on login page

jazdw, 2019-07-02:
If you have a user module that loads lots of small JS files on the login page or custom logo files etc then the rate limiter for the REST API kicks in and limits the requests resulting in HTTP 429 status error responses.

Solutions may include:
- Disable rate limiter for file store end points or provide separate settings
- Provide URL outside of /rest that the rate limiter does not limit
- Allow caching of file store files (currently the server sends cache-control headers which prevent caching)

terrypacker, 2019-07-12:
Also make note that the new endpoints should be considered in the Etag QoS and DoS filters as large files will cause problems.

jazdw, 2019-09-05:
@terrypacker Please elaborate, why does this have anything to do with the QoS and DoS filters? For the Etag filter I assume you mean the file store files should be treated as resources? Not really sure what you are getting at.

terrypacker, 2019-09-09:
@jazdw the QoS filter limits the maximum requests serviced at a time and the DoS filter limits the number of requests allowed per second and then throttles.  These settings are exposed in the env properties file so are configurable but I think we should consider that the endpoints to serve these files may still be within the control of the QoS and DoS filter.

terrypacker, 2019-09-19:
The DoS QoS PR https://github.com/infiniteautomation/ma-core-private/pull/28

jazdw, 2019-10-24:
Closed via infiniteautomation/ma-modules-public@d9417e0a5702fe0396182ddb7ee6c87d8d1fd28a

Added a filter which forwards from /file-stores/* to /rest/v2/file-stores/download-file/*

closed 2019-10-24 by jazdw