MangoAutomation/ma-core-public #1458: ETag Filter uses large amounts of memory and can cause OOM for REST/Web Requests

terrypacker, 2019-07-17:
When the MangoShallowEtagHeaderFilter or more specifically its base class the Spring ShallowEtagHeaderFilter executes it add various headers about cache control etc, but also wraps the response so that the response stream is not sent to the client, it is intercepted so it can then determine if the resource has changed by comparing the checksum of the stream to a previously requested one.  When done it copies the intercepted response body back into the original response and then proceeds.

This filter is currently mapped to /* for all requests.

Potential solutions:

1. The easiest is to skip the logic of the superclass entirely for REST requests.  This would still add some headers but not the Etag one and won’t ever set the response to 304 (not modified).  This solution would also solve the problem of trying to download a massive file from the filestore or any other rest endpoint that can generate a huge response.  

2. Another solution is to make a list of endpoints we don’t want to Etag and then match them in this filter. 

3. A better variation on this would be to create an annotation that we can add to Rest controllers or individual endpoints that tell the filter to not process them.  (Would require Mango 3.7) 

Solution 1 code in MangoShallowEtagHeaderFilter:
```
    @Override
    protected void doFilterInternal(HttpServletRequest request, HttpServletResponse response, FilterChain filterChain)
            throws ServletException, IOException {

        String header = null;

        if (restMatcher.matches(request)) {
            response.addHeader(HttpHeaders.CACHE_CONTROL, restHeader);
            filterChain.doFilter(request, response);
            return;
        } else if (resourcesMatcher.matches(request)) {
            String queryString = request.getQueryString();
            if (queryString != null && VERSION_QUERY_PARAMETER.matcher(queryString).find()) {
                header = versionedResourcesHeader;
            } else {
                header = resourcesHeader;
            }
        } else if (getMatcher.matches(request)) {
            header = defaultHeader;
        } else {
            header = NO_STORE;
        }

        response.addHeader(HttpHeaders.CACHE_CONTROL, header);

        super.doFilterInternal(request, response, filterChain);
    }
```

terrypacker, 2019-08-29:
This class now has a comment about removing for Mango 3.7 but I can't remember why and what the alternative solution is.  @jazdw we will need to discuss this again.

jazdw, 2019-09-05:
I would say that is because it is largely unnecessary and we could probably remove the Etag filter altogether.

As far as I can tell it is only really useful when requesting large resource files as it can save us re-transmitting the whole file and instead just calculate its hash. However we could just as easily use the [If-Modified-Since](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/If-Modified-Since) header and the file's last modified time.

- JSP pages are re-rendered and their hash checked so it is only saving us from transmitting a few kB of data
- REST requests are not cached at all `Cache-Control: no-cache`
- Our UI resource files all have a `?v=` query parameter and a large `max-age`

terrypacker, 2019-09-19:
@jazdw I am happy to remove this filter but you are referencing it in the UI module, can you remove those references and then I can remove it?

terrypacker, 2019-09-19:
Ah I figured out why it wasn't removed in 3.6, because of the references in the other modules.  The filter is in fact disabled in 3.7 already.

jazdw, 2019-09-20:
@terrypacker please link to commit where it was disabled.

terrypacker, 2019-09-20:
The general idea is the keep the logic from the MangoShallowETag filter's doFilter() method in terms of setting the caching headers.  

We also need to be sure that file resources get cached appropriately by the browser using the last-modified-time.  If this doesn't happen automatically we could potentially generate our own E-Tag using a last modified time and file size.

We need to check what will happen to the JSP pages and make adjustments if necessary.

terrypacker, 2019-09-24:
I added support for ETagging resource requests based on the MD5 hash of the last modified time and added tests for this in the api module: `etag.spec.js`

I've noticed that there are requests for the ui module that look like this that were not being handled correctly by the filter:
`/ui/fonts/MaterialIcons-Regular.woff2?v=8a9a261c8b8dfe90db11f1817a9d22e1`. @jazdw it looks like we will want to review the UIServlet in the UI module to verify what is being done manually via the cache control there as I've modified the new `MangoCacheControlHeaderFilter` to process the forwarded requests from that Servlet.

terrypacker, 2019-09-24:
See also this commit for the node tests: https://github.com/infiniteautomation/ma-modules-public/commit/ebfcb8fb1986c5f7110c2ce068c819358577b16f

terrypacker, 2019-10-18:
Closing as ETags have been replaced with cache control headers in Mango 3.7.0

closed 2019-10-18 by terrypacker

jazdw, 2019-10-24:
@terrypacker note that I added a configuration which will make the Jetty DefaultServlet use weak etags. It is disabled by default.
```
web.defaultServlet.etags=true
```