MangoAutomation/ma-core-public #1468: Publisher Enhancements Mango 3.7

terrypacker, 2019-07-30:
We will want to:

1.  break out the published points into their own table
2. Add a method to publisher RT that allows getting the status for a running publisher, then use this in a general purpose endpoint
3. Add additional information to the published point model about runtime status (will be a generic method on the PublisherRT/PublishedPointRT) and return this from a new query points endpoint.

terrypacker, 2019-08-29:
One complexity is that we already have 2 v2 REST controllers for publishers and both use models with a List of points in the publisher.

Proposed solution:

1. Add a new parameter to all `publisher-v2` endpoints that allows to optionally exclude the points from the model.

2.  Provide a new REST controller for publisher-points where one can access the points independent of being in a publisher model container.

jazdw, 2019-08-29:
>  I would also make the points field only included when it is non-null.
You mean the property in the JSON model? Yes I agree.

> Provide a new REST controller 
I think an endpoint in the existing controller would suffice.