MangoAutomation/ma-core-public #1469: Lazy point value initialization

Puckfist, 2019-07-31:
For systems with lots of points, the startup time came become pretty onerous. Time is primarily spent getting the latest values for all the points. This is necessarily to do if the point's runtime is going to start event detectors or interval logging with a certain point value, but for other points loading it more lazily may be an option, typified by the all data persistent receiver points.

terrypacker, 2020-09-29:
I've put some effort into this and have the following design:

1.  If we want to let the data sources initialize while Mango starts (which it currently does not @jazdw there is a wait for all to startup per group before moving on but may be a bug in that) we just need to make sure they start in the correct order as things like Meta points require other data sources to be started first.

2. When initializing the DataPointRT the main slowdown is the initialization of the cache in the constructor. There are 2 problems with this.  First the default cache size is 1 so the latest value is loaded.  Second even if you set the cache size to 0 it will expand to 1 when `DataPointRT.initialize()` is called because it wants the latest value.  I suspect we can delay the loading of the latest value and the subsequent initialization of the `toleranceOrigin` value in `DataPointRT.initialize()` until those fields are needed.

**OTHER PERFORMANCE NOTES**
- [ ] The PointValueDao uses system settings to purge values if all implementations of this were to be system setting listeners then we can remove the many calls to create a new one.

terrypacker, 2020-10-19:
Fixed by using LazyField in DataPointRT and the PointValueCache

closed 2020-10-19 by terrypacker