MangoAutomation/ma-core-public #1473: Event handler script validation doesn't include additional context objects

Puckfist, 2019-08-02:
When event handler scripts run, whether Set Point Handler or Email Handler, they are given the EventInstance as evt and the EventInstanceWrapper as event.

Email handlers are also given the whole FTL model under the key 'model', and UNCHANGED is wrapped in as 'CANCEL' for a way to prevent sending the email

terrypacker, 2019-08-13:
What are you suggesting as the solution?  To provide some dummy data in the model during validation?

Puckfist, 2019-08-13:
Yeah, that would be my first thought. TestEventInstance and TestEventInstanceWrapper

I think because we allow handlers on multiple event types that may be as good as it gets.

terrypacker, 2019-08-29:
There are 2 ways to solve this since the validation uses the common `/v2/script/validate` endpoint:

1.  Expose another field in MangoJavaScriptModel that allows the page to supply some 'additional context' objects to the script.  Then perform the logic on the page.

2.  Create a new endpoint for the validation of these and add the additional context objects in the backend.

terrypacker, 2019-09-20:
I'm leaning towards 1 as it would allow the UI developer to provide inputs for the dummy values in any context object so the user could test against desired data.

closed 2019-11-21 by terrypacker