MangoAutomation/ma-core-public #1484: Create disabled user functionality needs review

jazdw, 2019-09-03:
I can't see any other issues or pull requests pertaining to this new functionality. Here are some relevant commits -
a5b762db99b44290dafd62b66bcaeca0c638b935
8c2741868ca82caf2b592782e06b009ae6b64879

This ability for a non-admin user to create a new disabled user needs review. Specifically something I am concerned about is that it will reveal too much information about currently registered users. i.e. if I want to find out if a user with email `joe@something.com` is registered I can attempt to create a user with that email address and validation will fail if the someone already is using that email address.

jazdw, 2019-09-03:
I am also skeptical of this being a permission in the first place. If we want to enable this as core Mango functionality we should create a public endpoint for registering new users.

terrypacker, 2019-09-09:
@jazdw 

I agree that this could be done via a public endpoint.  The original use-case didn't require that and as such the security constraints were relaxed somewhat.  What is the process you were thinking?  Something like this:

1.  User visits public page, enters email address and clicks `register`
2. Mango then sends a registration email to that address with a link to register 
3. User goes to link and enters user settings, validates and saves user (which is disabled by default)
4. Admin must then review any self registered, disabled users and enable if desired.

The comment in this issue https://github.com/infiniteautomation/ma-dashboards/issues/291 implies that there will be another flag on a User to indicate that they are awaiting approval from an administrator.  The disabled flag serves this purpose in terms of how Spring Security views what a disabled account is so I think that should still be part of this process but I do agree that we need more information to distinguish what a self-registered disabled user is so it is easy for an administrator to find and enable them.

However I don't want to add another piece of logic into the authentication process that we need to check to ensure a user is able to login.  So I suggest that we add another flag to the user such as `selfRegistered` but only use this to filter/query users and let the disabled flag indicate if they can login.

So the proposed changes to a User are:

- [ ] add selfRegistered boolean field
- [ ] add dateCreated long field (instead of the proposed date self registered)
- [ ] add emailVerified boolean field that presently will only be set to true in the self registration process but can in the future be used to indicate a valid email address for a user.  This field isn't really required as the above process implies that a user has verified their email but I can see this as a useful feature in the future.

Thoughts?

terrypacker, 2019-09-09:
See also #1485 to add additional fields to user.

PR #1491 (2019-09-17): Public user registration
This fixes #1484 and #1485 

Summary of Work:

1. Upgrade28 to make email unique and add various columns to `users` table and use in Dao
2. Added Email Address Verification Service
3. Added Bean for system superadmin permission holder to use in backend when necessary
4. Wire in Users service to existing PasswordResetService and TokenAuthenticationService
5. Added env property to enable the public endpoints and adjusted the security context for this

See also:
https://github.com/infiniteautomation/ma-modules-public/pull/71
closed 2019-09-17 by jazdw