MangoAutomation/ma-core-public #1485: Add a role/position field to users

jazdw, 2019-09-03:
We have added a organization field to users, we also need a role/permission field.

One thing we need to decide with this is what to label this new field as. The user's page already displays an input labeled "Roles (permission groups)". Perhaps we should label it as "Organizational role" or "Role within organization"? Please comment @Joelhaggar @terrypacker

terrypacker, 2019-09-09:
I'll make the backend field named `organizationRole` and then the translation can be either of those mentioned above.  I'm leaning towards 'Role within organization', but either should be fine.

terrypacker, 2019-09-09:
Also add JSON field for custom info that can be assigned to a user.

closed 2019-11-21 by terrypacker