MangoAutomation/ma-core-public #1495: Add some useful value monitors such as CPU usage and disk space

jazdw, 2019-09-16:
These values would be useful values to show on the "Internal metrics" page and to be able to setup as internal data source datapoints -

- Java CPU usage
- Total CPU utilization
- System load
- Available disk space (for MA_HOME, file store, DB location, time series DB location etc)

I believe there are some limited Java APIs to get this information but this project may be useful -
https://github.com/hyperic/sigar/wiki/overview
It also supports getting network interfaces etc so it may be useful for the MangoES configuration module.

terrypacker, 2019-10-30:
That library doesn't seem to support ARM unless I'm missing something...

We can get CPU utilization already from the VMStat data source (On linux only)

Disk space we should be able to work out using this:

```
File cDrive = new File("C:");
System.out.println(String.format("Total space: %.2f GB",
  (double)cDrive.getTotalSpace() /1073741824));
System.out.println(String.format("Free space: %.2f GB", 
  (double)cDrive.getFreeSpace() /1073741824));
System.out.println(String.format("Usable space: %.2f GB", 
  (double)cDrive.getUsableSpace() /1073741824));
```
But we would need to account for multiple drives and locations.  I could see this being a type of data point on the Internal Metrics data source where you can choose the location to measure the total/free/usable metrics.

terrypacker, 2019-10-30:
I like the preconfigured options idea, here are the locations we need to monitor:
- [ ] MA_HOME
- [ ] filestore
- [ ] all filestore definitions should optionally be able to provide a monitoring value for their size
- [ ] SQL database
- [ ] logs directory
- [ ] backup directory
- [ ] NoSQL point value directory
- [ ] NoSQL data directory (non point value data for points)

There is a module element definition that is exposed via `GET /rest/v2/server/system-info` that provides a lot of information like this that I don't think made it into the UI.  I would suggest we create MonitoredValue classes for all of these so we can track their values as data points.  The definition is called `SystemInfoDefinition`

closed 2019-11-21 by terrypacker