MangoAutomation/ma-core-public #1506: Event hander - can't utilize mailing lists

paul-geyer-moser, 2019-10-24:
Creating and Editing the mailing list seems to be working fine - it is when the mailing list is utilized with an event handler that we have an issue.

**When using the MangoUI Event Handlers page**
- Selecting a mailing list from the "select mailing list" dropdown and saving it throws an error:
	
Error saving event handler: Bad Request — com.fasterxml.jackson.databind.exc.InvalidTypeIdException: Could not resolve type id 'MAILING_LIST' as a subtype of [simple type, class com.infiniteautomation.mango.rest.v2.model.mailingList.EmailRecipientModel]: known type ids = [ADDRESS, USER] (for POJO property 'activeRecipients') at [Source: (PushbackInputStream); line: 1, column: 2225] (through reference chain: com.infiniteautomation.mango.rest.v2.model.event.handlers.EmailEventHandlerModel["activeRecipients"]->java.util.ArrayList[0])

**When using the Legacy UI Event Handlers page:**
- It allows me to select the Mailing list and save.  It returns the "Event Handler has been saved" message.

The repercussions:
- Legacy UI - On page refresh,etc.  It shows the mailing list.
- MangoUI - It now shows an "Empty Pill box" in the list of valid user ids that can be x-ed out (removed).  Without deleting the "empty user" and saving - it throws this error:

Error saving event handler: Bad Request — com.fasterxml.jackson.databind.exc.InvalidTypeIdException: Could not resolve type id 'MailingListEntryModel' as a subtype of [simple type, class com.infiniteautomation.mango.rest.v2.model.mailingList.EmailRecipientModel]: known type ids = [ADDRESS, USER] (for POJO property 'activeRecipients') at [Source: (PushbackInputStream); line: 1, column: 472] (through reference chain: com.infiniteautomation.mango.rest.v2.model.event.handlers.EmailEventHandlerModel["activeRecipients"]->java.util.ArrayList[2])

Removing this "empty user" - it can save normally in MangoUI now.

**Looking at the data in REST API  -** 
**event-handlers-rest-controller  (GET) /event-handlers/{xid}**
returns null xid for the event handler for activeRecipients of recipientType "MailingListEntryModel":

"activeRecipients": [
    {
      "recipientType": "USER",
      "username": "xxx.yyy"
    },
    {
      "recipientType": "USER",
      "username": "abc.123"
    },
    {
      "recipientType": "MailingListEntryModel",
      "xid": null
    }
  ],

jazdw, 2019-10-24:
Thankyou for the detailed report @paul-geyer-moser

terrypacker, 2019-10-25:
@jazdw I've fixed this in the REST api but there are a few UI bugs:

1.  missing translation when you add a list, the icon tooltip shows 
```
!!ui.components.emailRecipient!!
```
2.  When you load a saved email event handler the name of the mailing list is not on the chip component, when initially add it before saving it the name is shown on the chip.  The name is definitely in the model returned on the REST api so I think it is just not added to the chip properly.

I intentionally left off the RELEASE NOTE commit as this may be cherry picked over and released for 3.6.x next week.

jazdw, 2019-10-25:
Still a REST API bug.
POST body -
[image]

Response -
[image]

terrypacker, 2019-10-29:
@jazdw I could supply the name with the Mailing list in addition to the XID if you would rather put that on the chip?

terrypacker, 2019-11-01:
@jazdw the name is now part of the model I will close this after you add that into the UI.

jazdw, 2019-11-01:
@terrypacker should not require any changes to the UI. Unless I am missing something? Did you test it?

terrypacker, 2019-11-01:
All good, just checked and it works as expected.

closed 2019-11-01 by terrypacker