MangoAutomation/ma-core-public #1510: Jetty/Spring Error Handlers

terrypacker, 2019-11-01:
There are 2 parts to this issue, the first is fixing the generic Jetty error handler we have called MangoErrorHandler in the core, the second is fixing the RestExceptionHandler classes in the api.

1. MangoErrorHandler - is causing an IllegalStateException due to the response being committed before getting to this handler.  

See https://gist.github.com/jazdw/d6b1838ad9776c42ae60d7f50b6959ab
 
2. The RestExceptionHandlers are not redirecting properly and have references to JSP pages in them.

jazdw, 2019-11-05:
@terrypacker I am subscribed to the Jetty Users mailing list and saw this thread which might be relevant - https://www.eclipse.org/lists/jetty-users/msg09189.html

terrypacker, 2019-11-06:
Something else to note is that our MangoErrorHandler is being called on the 401 response from the login page when /current is being polled by the page.  I'm not sure this should happen and if so then we need to improve what the MangoErrorHandler is doing for the various error responses.

My feeling is that this should have already be handled at the REST level...

closed 2020-12-04 by terrypacker