MangoAutomation/ma-core-public #1513: H2Proxy fails with certain H2 database names

jazdw, 2019-11-12:
e.g. given
```
ls -l /opt/mango/databases/
total 860
-rw-r--r--  1 mango mango 4870144 Nov 12 14:29 mango.199.h2.db
-rw-r--r--  1 mango mango    1790 Nov 12 14:29 mango.199.trace.db
drwxr-xr-x 19 mango mango    4096 Nov 12 14:29 mangoTSDB
drwxr-xr-x  2 mango mango    4096 Nov 12 14:29 mangoTSDBAux
```

H2Proxy fails -
```
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: Exception in thread "high-pool-2-thread-2" java.lang.NullPointerException
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.CoreLicenseDefinition$2.run(CoreLicenseDefinition.java:161)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.timer.sync.Synchronizer$TaskWrapper.run(Synchronizer.java:150)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.timer.Task.runTask(Task.java:179)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.timer.TaskWrapper.run(TaskWrapper.java:23)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at java.base/java.lang.Thread.run(Thread.java:830)
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: ERROR 2019-11-12T15:45:25,424 (com.serotonin.m2m2.Main.main:146) - Error during initialization
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: java.lang.ArrayIndexOutOfBoundsException: Index -2 out of bounds for length 1
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.db.H2Proxy.getVersion(H2Proxy.java:445) ~[mango-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.db.H2Proxy.getVersion(H2Proxy.java:438) ~[mango-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.db.H2Proxy.getUrl(H2Proxy.java:195) ~[mango-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.db.H2Proxy.initializeImpl(H2Proxy.java:66) ~[mango-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.db.AbstractDatabaseProxy.initialize(AbstractDatabaseProxy.java:71) ~[mango-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.Lifecycle.databaseInitialize(Lifecycle.java:760) ~[ma-priv-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:312) ~[ma-priv-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: #011at com.serotonin.m2m2.Main.main(Main.java:143) ~[ma-priv-3.6.5.jar:?]
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: INFO  2019-11-12T15:45:25,425 (com.serotonin.m2m2.Lifecycle.terminate:427) - Mango Lifecycle terminating...
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: INFO  2019-11-12T15:45:25,504 (com.infiniteautomation.nosql.MangoNoSqlProxy.shutdown:115) - Terminating NoSQL Batch Write Manager.
Nov 12 15:45:25 mangogateway start-mango.sh[2374]: INFO  2019-11-12T15:45:25,540 (com.serotonin.m2m2.Lifecycle.terminate:578) - Mango Lifecycle terminated.
```

We should be doing better Regex matching on the filenames to find the actual DB files.

terrypacker, 2019-11-19:
@jazdw it was the filtering of the files to check that needed improvement.  Now any potential db must
1. have the extension of `.db` (added by H2 when created, nothing we can control)
2. be a file (not a directory)
3. start with the name of the db in the url

terrypacker, 2019-12-10:
Closing as released and tested.

closed 2019-12-10 by terrypacker