MangoAutomation/ma-core-public #1514: Event Handler Permissions

terrypacker, 2019-11-20:
Event handlers currently do not have any permissions.  Should we add some to them and then enforce them in the EventHandlerService?  Currently only superadmin can read/create/edit them.

jazdw, 2019-11-21:
I think something like a set point event handler should be able to be used by a wider range of people for sure, also email handlers. We probably should have a permission that controls if a user is allowed to create each type, and a per handler edit permission of course.

jazdw, 2019-11-21:
We should be careful about allowing non-admin users to create email handlers with scripts however.

Joelhaggar, 2019-11-21:
I think we will want to include this discussion with the permission system overhaul as part of Mango 4.0 for stallion.

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker