MangoAutomation/ma-core-public #1518: NPE getting persistent publisher status

jazdw, 2019-12-09:
**Describe the bug**
NPE when trying to retrieve a persistent publishers's status via the legacy UI. The new UI does not yet have this facility.

**To Reproduce**
Click the get status button. This is on an unlicensed instance of Mango and the publisher is enabled. It looks like the lack of license stops the publisher from starting so the RT does not exist.

**Mango Version (please complete the following information):**
 - Core Version: 3.8.x SNAPSHOT
 - Module Version 3.8.x SNAPSHOT

**ma.log Snippet**
```
ERROR 2019-12-09T18:32:33,153 (com.serotonin.m2m2.web.dwr.util.ExceptionDetectionFilter.doFilter:38) - DWR invocation exception
java.lang.NullPointerException: null
        at com.serotonin.m2m2.persistent.pub.PersistentSenderRT.getConnectingIndex(PersistentSenderRT.java:154) ~[?:?]
        at com.serotonin.m2m2.persistent.dwr.PersistentPublisherDwr.getPersistentSenderStatus(PersistentPublisherDwr.java:280) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
        at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
        at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
        at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
        at org.directwebremoting.impl.ExecuteAjaxFilter.doFilter(ExecuteAjaxFilter.java:34) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
        at com.serotonin.m2m2.web.dwr.util.TranslationsFilter.doFilter(TranslationsFilter.java:37) ~[mango-3.8.0-SNAPSHOT.jar:?]
        at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
        at com.serotonin.m2m2.web.dwr.util.ExceptionDetectionFilter.doFilter(ExceptionDetectionFilter.java:26) ~[mango-3.8.0-SNAPSHOT.jar:?]
        at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
        at com.serotonin.m2m2.web.dwr.util.DwrPermissionFilter.doFilter(DwrPermissionFilter.java:46) ~[mango-3.8.0-SNAPSHOT.jar:?]
        at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:431) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:283) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.servlet.PlainCallHandler.handle(PlainCallHandler.java:52) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.servlet.UrlProcessor.handle(UrlProcessor.java:101) ~[dwr-2.0.11.jar:?]
        at org.directwebremoting.servlet.DwrServlet.doPost(DwrServlet.java:146) ~[dwr-2.0.11.jar:?]

```

terrypacker, 2019-12-10:
@jazdw next time can you put the issue in the module's repo.

Also I am now terminating the publisher if it is started without a license instead of letting it run but not connect. It raises an event about the problem first.

 We could add some validation when saving the publisher that says you are unlicensed but I didn't do that yet because I'm not sure what the context key would be. I guess I could just pick one of the fields like name or something.  Thoughts?

jazdw, 2019-12-10:
> next time can you put the issue in the module's repo.
The problem with doing this is that it is not visible to our users.

> We could add some validation when saving the publisher that says you are unlicensed
Pretty sure this already happens. I couldn't save the publisher with enabled set to true but I could use the enable-disable API.

>  I'm not sure what the context key would be
You can also set generic messages without a context key.

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker