MangoAutomation/ma-core-public #1520: Review hex conversion in AnalogRenderer

jazdw, 2019-12-20:
@terrypacker you added this code here - https://github.com/infiniteautomation/ma-core-public/commit/2f43e7006c8a5d381f55eeff11e62418a29d80c9

Not sure where this is used or how exactly it works so please review that this commit did not break it -
https://github.com/infiniteautomation/ma-core-public/commit/92b4ac27f1baf209908bdef87cc38af3bcc53080

jazdw, 2019-12-21:
This method might be a better fit - java.math.BigInteger.BigInteger(String, int)

terrypacker, 2020-10-28:
@jazdw not sure if you were referring to the convert `into hex` or `from hex` portion of that code.  

For `into hex` we could use something like this but we still need to trim the digits based on the format:
```
String raw = new BigInteger(Double.toString(value)).toString(16);
```

jazdw, 2020-10-29:
https://github.com/infiniteautomation/ma-core-public/commit/92b4ac27f1baf209908bdef87cc38af3bcc53080#diff-c194df6260e168bcd391d06a2e01e758505707809dc220c6f82059b7cb4c6ed7R142

terrypacker, 2020-10-29:
Just found that I cannot get that method to execute because the UI won't let me enter `0x00` and set a Numeric point.  Not that it would matter as the backend REST api doesn't expect a string for that data type yet.

commit ba7fc472 (2020-10-29): Use BigDecimal to parse string for hex. Fixes #1520
closed 2020-10-29 by terrypacker