MangoAutomation/ma-core-public #1522: REST Controllers v2 Conversion

terrypacker, 2020-01-14:
This is the list of REST controllers and UI code that will be ported to v2 for Mango 4.0

Excel Reports
- [ ] GET `/rest/v1/excel-report-templates/email-templates`
- [ ] GET `/rest/v1/excel-reports/download/{xid}`
- [ ] GET `/rest/v1/excel-reports/view/{xid}`

AuditRestController
- [ ] GET `/rest/v1/audit`

DeviceNameRestController
- [ ] GET `/rest/v1/device-names`
- [ ] GET `/rest/v1/device-names/by-data-source-id`
- [ ] GET `/rest/v1/device-names/by-data-source-xid`

MangoIO Tools
- [ ] GET `/rest/v1/data-points/' + dataPoint.xid`
- [ ] GET `/rest/v1/data-points/data-source/' + dataSourceXid`
- [ ] All endpoints in v1 MangoIO controller
- [ ] POST `/rest/v1/modbus/serial/locator`
- [ ] POST `/rest/v1/modbus/serial/write`
- [ ] POST `/rest/v1/modbus/serial/read`
- [ ] 

ExampleRestControllerv2
- [ ] move all methods to TestRestController

JsonDataRestController
- [ ] fix v1 tests to pass in v2

LoggingRestController
- [ ] fix v1 tests to pass in v2

ModulesRestController
- [ ] ensure v2 controller has all the same endpoints

Point Value Rest Controller
- [ ] PUT /rest/v1/point-values/{xid} no v2 variant

RuntimeManagerRestController
- [ ] no v1 tests need v2 tests

ThreadsRestController
- [ ] move endpoints to server v2 controller

UI 
- [ ] scriptingEditor.html contains a reference to a v1 explain endpoint for data points
- [ ] `/v1/logging` needs to be converted to v2
- [ ] `/v1/modules` needs to be converted to v2
- [ ] `v1/json-data` needs to be converted to v2
- [ ] `/rest/v1/device-names` needs to be converted to v2
- [ ] `/rest/v1/audit` needs to be converted to v2
- [ ] PUT `/rest/v1/point-values` needs to be converted to v2
- [ ] `/rest/v2/realtime` has been removed for 4.0
- [ ] `/rest/v1/system-metrics` needs to be converted to v2
- [ ] `/rest/v1/system-settings` needs to be converted to v2
- [ ] `/rest/v1/threads` moved to `/rest/v2/server/threads`
- [ ] `/rest/v1/translations` needs to be converted to v2
- [ ] `/rest/v1/comments` needs to be converted to v2
- [ ] `/rest/v1/users` needs to use v2 and update to pickup model changes
- [ ] `/rest/v1/watch-lists` needs to be converted to v2
- [ ] `/rest/v1/work-items` needs to be converted to v2 

WatchlistRestController
- [ ] add v2 tests

Stallion
- [ ] POST `/rest/v1/events/module-defined-query` for Maintenance Events will need to change URL to post to new MaintenanceEvents controller 
- [ ] GET `/rest/v2/realtime` this has been removed and we will need an alternate solution

Maintenance Events
- [ ] POST `/rest/v1/events/module-defined-query` for Maintenance Events will need to change URL to post to new MaintenanceEvents controller

jazdw, 2020-01-22:
I have added auto generated tests for the following v1 APIs:
- audit-rest-controller
- device-name-controller
- json-data-rest-controller
- logging-rest-controller
- modules-rest-controller
- runtime-manager-rest-controller
- system-metrics-rest-controller
- system-settings-rest-controller
- thread-monitor-rest-controller
- translations-rest-controller
- user-comment-rest-controller
- work-item-rest-controller

jazdw, 2020-01-27:
V2 events issues that need to be addressed / or comments added to release notes about changes:
- [ ] Can no longer query on dataPointId
- [ ] Model eventType no longer has a dataPointId property (but confusingly still has a dataSourceId property)
- [ ] Model no longer has an active property

terrypacker, 2020-01-28:
@jazdw as for the watchlist monitor point, the dao changed packages and that needed to be updated in the Internal data source module. Keep an eye out for others that are missing, I did a quick check and didn't see any others yet.

terrypacker, 2020-01-28:
@jazdw 

> Model eventType no longer has a dataPointId property (but confusingly still has a dataSourceId property)

This is how the new event types in v2 work.  The data point event type as a data source ID in addition to the reference1 = data point and reference2 = point event detector.  We designed the event types endpoint to work this way.

jazdw, 2020-01-28:
> This is how the new event types in v2 work. The data point event type as a data source ID in addition to the reference1 = data point and reference2 = point event detector. We designed the event types endpoint to work this way.

Ah I was thinking that ref2 was the dataSourceId. That makes more sense.

terrypacker, 2020-01-29:
Thanks, we removed that module and Reports.  So these 2 will also be gone: 

`internal_mango_num_reports`
`internal_mango_num_report_instances`

jazdw, 2020-01-29:
Still seeing the duplicate events come through on the WebSocket and on the REST API, same timestamp, message etc but different ID.
[image]

terrypacker, 2020-01-31:
The script context variable model that was previously used in v1 is gone and the pre-existing v2 model is slightly different, the `dataPointXid` field is now `xid`.  We need to update the UI code for these 2 modules:

- [ ]  Meta Data Source (Point locator)
- [ ]  Scripting Data Source

terrypacker, 2020-02-04:
- [ ] User permissions are a Set<String> in v2.  @jazdw The model existed before the upgrade so I think we will need to modify the UI.

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker