MangoAutomation/ma-core-public #1523: Make upgrade/install Module Lifecycle booleans more reliable.

terrypacker, 2020-01-24:
In 4.0 we can leverage the installedModule table.

terrypacker, 2020-01-24:
See #1027

terrypacker, 2020-03-06:
Make sure that we also do this for any Core installed module elements.  Their lifecycle methods are NOT called currently...

terrypacker, 2020-03-10:
@jazdw this is ready for your review.  The only thing would be to confirm that we want to pass in the previous version and current version or do we want to provide other information like previously with the upgrade/install

jazdw, 2020-03-16:
@terrypacker you removed the install / update booleans from the module hooks. How do we now tell if we are installing / upgrading? I assume we should check the module's `previousVersion` property? Please confirm.

Can you also give some more details on what you changed in order to make this more reliable? Are you using the versions from the SQL table? What about hooks before the DB is started?

terrypacker, 2020-03-16:
@jazdw this is the logic we now use to determine if a new module is being installed/upgraded.  I'm open to reworking this if you have any better ideas:

```
    @Override
    public void postDatabase(Version previousVersion, Version current) {
        if(previousVersion == null) {
            install();
        }else if(current.greaterThan(previousVersion)) {
            upgrade();
        }
    }
```

The versions are determined from the `installedModules` database table so they are only available after the database is initialized.  This seems to be fine as there was no code that needed to perform install/upgrade logic before this part of the lifecycle.  There is only 1 lifecycle hook before the database is started and it does not have any version information in it now.

```
    /**
     * Called immediately after the module is loaded, before the system is initialized.
     */
    public void preInitialize() {
        // Override as required
    }
```

jazdw, 2020-03-16:
OK great. So preInitialize() will not have access to version information, it will be null?

terrypacker, 2020-03-16:
There are no version parameters in that hook.  They are passed into the other hooks.

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker