MangoAutomation/ma-core-public #1524: Report Generation: freemarker.core.NonStringOrTemplateOutputException

paul-geyer-moser, 2020-02-03:
I am creating a new automated report that ones once daily w/ the previous days info.  I can create the report just fine.  It never emails daily and when running it on-the-fly it throws an exception:

The datasource is an snmp datasource with several datapoints.  All of them are saved as Alphanumeric.

**Server error**
freemarker.core.NonStringOrTemplateOutputException: For "${...}" content: Expected a string or something automatically convertible to string (number, date or boolean), or "template output" , but this has evaluated to an extended_hash (wrapper: f.t.SimpleHash):
==> point.tags  [in template "Cell Modem info" at line 121, column 56]

----
FTL stack trace ("~" means nesting-related):
	- Failed at: ${point.tags}  [in template "Cell Modem info" at line 121, column 54]
----
	at com.serotonin.m2m2.reports.web.ReportChartCreator.createContent (ReportChartCreator.java:266)
	at com.serotonin.m2m2.reports.web.ReportChartHandler.handleRequest (ReportChartHandler.java:36)
	at com.serotonin.m2m2.web.mvc.UrlHandlerController.handleRequest (UrlHandlerController.java:36)
	at org.springframework.web.servlet.mvc.SimpleControllerHandlerAdapter.handle (SimpleControllerHandlerAdapter.java:52)
	at org.springframework.web.servlet.DispatcherServlet.doDispatch (DispatcherServlet.java:1039)
	at org.springframework.web.servlet.DispatcherServlet.doService (DispatcherServlet.java:942)
	at org.springframework.web.servlet.FrameworkServlet.processRequest (FrameworkServlet.java:1005)
	at org.springframework.web.servlet.FrameworkServlet.doGet (FrameworkServlet.java:897)
	at javax.servlet.http.HttpServlet.service (HttpServlet.java:687)
	at org.springframework.web.servlet.FrameworkServlet.service (FrameworkServlet.java:882)
	at javax.servlet.http.HttpServlet.service (HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle (ServletHolder.java:865)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter (ServletHandler.java:1655)
	at org.springframework.web.filter.CharacterEncodingFilter.doFilterInternal (CharacterEncodingFilter.java:200)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter (OncePerRequestFilter.java:107)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter (ServletHandler.java:1642)
	at com.infiniteautomation.mango.webapp.filters.WebContextFilter.doFilter (WebContextFilter.java:59)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter (ServletHandler.java:1634)
	at com.infiniteautomation.mango.webapp.filters.MangoCacheControlHeaderFilter.doFilter (MangoCacheControlHeaderFilter.java:155)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter (ServletHandler.java:1642)
	at com.infiniteautomation.mango.webapp.filters.UrlSecurityFilter.doFilter (UrlSecurityFilter.java:204)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter (ServletHandler.java:1642)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:320)
	at org.springframework.security.web.authentication.switchuser.SwitchUserFilter.doFilter (SwitchUserFilter.java:200)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke (FilterSecurityInterceptor.java:127)
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter (FilterSecurityInterceptor.java:91)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at com.serotonin.m2m2.web.mvc.spring.security.RateLimitingFilter.doFilterInternal (RateLimitingFilter.java:53)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter (OncePerRequestFilter.java:107)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at com.serotonin.m2m2.web.mvc.spring.security.PermissionExceptionFilter.doFilter (PermissionExceptionFilter.java:34)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter (ExceptionTranslationFilter.java:119)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.session.SessionManagementFilter.doFilter (SessionManagementFilter.java:137)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter (AnonymousAuthenticationFilter.java:111)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter (SecurityContextHolderAwareRequestFilter.java:170)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.session.ConcurrentSessionFilter.doFilter (ConcurrentSessionFilter.java:155)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.authentication.AbstractAuthenticationProcessingFilter.doFilter (AbstractAuthenticationProcessingFilter.java:200)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.authentication.AbstractAuthenticationProcessingFilter.doFilter (AbstractAuthenticationProcessingFilter.java:200)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.authentication.logout.LogoutFilter.doFilter (LogoutFilter.java:116)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.csrf.CsrfFilter.doFilterInternal (CsrfFilter.java:100)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter (OncePerRequestFilter.java:107)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.header.HeaderWriterFilter.doFilterInternal (HeaderWriterFilter.java:74)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter (OncePerRequestFilter.java:107)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.context.SecurityContextPersistenceFilter.doFilter (SecurityContextPersistenceFilter.java:105)
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter (FilterChainProxy.java:334)
	at org.springframework.security.web.context.request.async.WebAsyncManagerIntegrationFilter.doFilterInternal (WebAsyncManagerIntegrationFilter.java:56)
[... 70 lines cut ...]
          "pointKey": "cradlepoint",
          "weight": 1,
          "consolidatedChart": true,
          "pointXid": "DP_ae055bff-f52e-4ae4-b561-c5303cfe8d96"
        }
      ],
      "schedule": true,
      "xid": "REPORT_2aea16a3-b747-476e-a6d0-9af6bae3ddbb",
      "runDelayMinutes": 0,
      "recipients": [
        {
          "recipientType": "ADDRESS",
          "address": "userone@test.com"
        }
      ],
      "dateRangeType": "RELATIVE",
      "name": "Cell Modem info",
      "pastPeriodType": "DAYS",
      "user": "user.one",
      "schedulePeriod": "DAY",
      "email": true,
      "includeEvents": "NONE"
    }
  ]
}

terrypacker, 2020-02-03:
@paul-geyer-moser your FTL template has a bug in it. You are trying to output a HashSet as a String.  You likely will need to iterate over that and build the output.

terrypacker, 2020-02-14:
Closing as no feedback, assume not a bug.

closed 2020-02-14 by terrypacker