MangoAutomation/ma-core-public #1525: Mango 4.0 Query Optimization

terrypacker, 2020-02-07:
For Mango 4.0 I have identified these RQL queries from the UI that will need optimization:

- [ ] Watchlist loading of points for new list https://localhost:8443/rest/v2/data-points?limit(10000) the bulk of the time is spent loading the tags via loadRelationalData(). Do we really need 10,000 points here? Pagination?

- [ ] Data source point tab https://localhost:8443/rest/v2/data-points?eq(dataSourceXid,tuna-melt8-1000-points)&limit(100000) 100k?  Should pagenate.

- [ ] Data Sources https://localhost:8443/rest/v2/data-sources?limit(1000) What if you have > 1000 data sources?

- [ ] Event Handlers listing data point event types https://localhost:8443/rest/v2/event-types/DATA_POINT/null should implement pagination on backend AND frontend.  We use that query object list style processing so this will take some thought.

- [ ] Event Handlers listing data source event types https://localhost:8443/rest/v2/event-types/DATA_SOURCE/null should implement pagination on backend AND frontend. We use that query object list style processing so this will take some thought.

- [ ] Bulk data point edit https://localhost:8443/rest/v2/device-names should implement a specific query and not get the entire data point

terrypacker, 2020-02-19:
@jazdw some of these should be simple UI changes to use a different limit, others might take sometime to use pagination if it isn't supported in the used component. 

Can you take a look at them.  Feel free to edit the description to add more info or make individual issues that I can pass on to our other developers.

jazdw, 2020-02-25:
Data Source query for points like `/rest/v2/data-points?eq(dataSourceXid,tuna-melt8-1000-points)&limit(100000)` is done because I need all the data points for the data source in order to do the bulk edit functions.

Is the implicit limit of 100 still present for all REST queries?

terrypacker, 2020-02-26:
Yes, but I would consider removing that...

https://github.com/infiniteautomation/ma-core-public/blob/35e6b8a2984c70da36f275dd9f80c5738402d6fa/Core/src/com/infiniteautomation/mango/db/query/ConditionSortLimit.java#L27

terrypacker, 2020-03-06:
I took a crack at improving  http://localhost:8443/rest/v2/users?sort(username,id)&limit(50,50) by executing a count on the role mapping table to see if the user has any roles in a permission.  This did slightly improve performance.  However I think these queries are all fast but the fact that we make a query for every permission for every user then we end up making 50 * 45 = 2,250 queries to get only 50 users.  I think we are going to have to cache the roles for a permission definition inside the definition.  When I fix up this issue I'll re-visit this idea: #1523

terrypacker, 2021-03-08:
Closing and will re-open as new performance issues are found.  Most of these queries have changed since this was open.

closed 2021-03-08 by terrypacker