MangoAutomation/ma-core-public #1531: Verify session persistence works correctly

jazdw, 2020-02-19:
There is a new setting `sessionCookie.persistent=true` which will enable the persistence of sessions to disk. We need to verify that it works correctly and as intended. In particular the serialization of users and roles which are stored onto the disk and the implications of this.

#1346 is related to this.

terrypacker, 2020-02-19:
Default Session Cache configuration options:
https://www.eclipse.org/jetty/documentation/current/session-configuration-sessioncache.html

Referenced commits:

5dd1baee7279e3e7c2de963627abb4686966b905 - serialize role, user properly
07cd92555ca48d44e65e1064fb3bdda17bf73c05 - serialize translatable messages

terrypacker, 2020-02-20:
The validation result must be getting into the session as I get an exception about the session being un-writeable.

```
WARN  2020-02-19T15:52:07,419 (org.eclipse.jetty.server.session.SessionHandler.complete:372) -  
org.eclipse.jetty.server.session.UnwriteableSessionDataException: Unwriteable session node0fnht08yvbhdx1hsasb1oly1z91 for node0__0.0.0.0
... [ much extra stuff left out]
Caused by: java.io.NotSerializableException: com.serotonin.m2m2.i18n.ProcessResult
```

@jazdw I added this commit 8dfe4aecf837e5f099d7f063cf477ce839a87b39 thinking it might solve my problem but I am confident this is not a good solution after playing with it.  It was reverted here 05b3b973185e0aa211ef0b10d50676f5537c70ab.

The issue is that the ProcessResult contains a map:
`private Map<String, Object> data = new HashMap<String, Object>();`

with no restriction on serializability of the values.

We will need to find a way to filter what gets saved, or if that is not possible then we will need to be quite careful with this as once this happens Jetty just keeps trying to save the session and spitting out errors about that being in the session.  The session file is also gone, which I guess is good.

terrypacker, 2020-02-20:
I've removed the data map from the process result but have not yet serialized the ProcessResult or ProcessMessage as I'm not sure we really want to put exceptions into the session.

terrypacker, 2020-02-21:
I've come full circle on this and decided that since anything that extends Exception must implement Serializable we should be able to store the exception in the session.    One idea is to test via pre-serializing the exception before we put it into the session just in case that fails we can skip putting it into the session.

terrypacker, 2020-02-21:
I found that the User.id property is -1 when it is deserialized back into the session, which makes sense as that property is NOT serialized by our custom code.

jazdw, 2020-02-21:
Yeah I'm not sure we want to serialize everything in the session attributes onto disk. I would be happier probably if we just serialized the user ID and loaded the user from the DB on restart or something like that. I am not sure we need to keep anything from the attributes. We might need a custom data store implementation or something to do this though.

It also seems to update the session files quite rapidly right now. I am not sure why, perhaps because a session attribute is being updated rapidly?

terrypacker, 2020-02-26:
Reference commit to only save pieces of the session.  We may want to strip out all the attributes and only specifically save what we want.  Currently this only removes the few attributes I know are problematic.

https://github.com/infiniteautomation/ma-core-public/commit/9bed0aab33aea1852fc9be1b5f0ae5c6f67b7a3c

jazdw, 2020-05-04:
We should switch this from disk persistence to use a SQL table and jOOQ.

terrypacker, 2020-05-05:
I have added the L2 cache using Caffeine but am not sure that it gets us much after looking at this further as there is an L1 cache that sits in front of our JDBC store already.

terrypacker, 2020-05-06:
I have added some test helper endpoints when we are ready to add tests to the session.spec.js test.

 infiniteautomation/ma-modules-public@fb605b17af9221b188a60d136a00b8ce45dc11c1

We should test that expiring a User's session will also expire any persistent sessions

terrypacker, 2020-05-08:
The last thing to look at is how often the session is being persisted.  It appears that every request is calling finish on the session which persists it.  We don't want to load the db.

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker