MangoAutomation/ma-core-public #1542: Upgrade Spring to version 5.2

jazdw, 2020-03-01:
Does not look like it should be too much work
https://github.com/spring-projects/spring-framework/wiki/What's-New-in-Spring-Framework-5.x#whats-new-in-version-52
https://github.com/spring-projects/spring-framework/wiki/Upgrading-to-Spring-Framework-5.x#upgrading-to-version-52

terrypacker, 2020-05-05:
`@RequestMapping()` no longer matches all paths.  This will be important for our RQL endpoints.  They will all need to be replaced with `@RequestMapping("/**")`

terrypacker, 2020-05-05:
This seems like the only other item that may affect us:

* CORS processing is now only used for CORS-enabled endpoints
* CORS processing for skipped for same-origin requests with an Origin header
* Vary headers are added for non-CORS requests on CORS endpoints

terrypacker, 2020-05-05:
Closing as done.  Opened #1575 to test CORS.

closed 2020-05-05 by terrypacker