MangoAutomation/ma-core-public #1545: Add Log location/tool to UI for all modules that use this setting

terrypacker, 2020-03-03:
Any Data Source that has logging inputs should display the location of the log file and a way to view it.  This will use the concept from #319

Modules that require this change:
- [ ] Scripting data source
- [ ] Meta data source
- [ ] ssh data source
- [ ] modbus data source
- [ ] serial data source
- [ ] ssh data source
- [ ] tcp ip data source

jazdw, 2020-03-03:
You should link to the logging console page using a button or `<a>` element with a `ui-state` attribute like this - `ui-state="'ui.settings.systemStatus.loggingConsole'" ui-state-params="{logFileFilter: 'ma.log', displayLogFile: 'ma.log'}"`

jazdw, 2020-03-04:
These are implemented now 
[image]
[image]

terrypacker, 2020-03-17:
Confirmed all data sources look/work as expected.

closed 2020-03-17 by terrypacker