MangoAutomation/ma-core-public #1552: Use Read/Edit permission

terrypacker, 2020-03-05:
Add Read and Edit permission to User. This will allow letting certain users edit other users and view other users.

terrypacker, 2020-04-16:
Read view will restrict what fields can be seen.
Edit should be restricted to what fields they can edit.

terrypacker, 2020-08-20:
Check the UsersViewPermissionDefinition when we implement this as it is not used anywhere anymore but could be used in the UI to hide/show the Users/Roles menu item.

jazdw, 2021-03-01:
@terrypacker do you think we should remove the `permissions.user.editSelf` system permission? I would imagine so. Do you think we can/should do an upgrade on this?

jazdw, 2021-03-01:
Also @terrypacker can you please elaborate on what you mean by these -

> Read view will restrict what fields can be seen.
Edit should be restricted to what fields they can edit.

terrypacker, 2021-03-01:
@jazdw 

> Read view will restrict what fields can be seen.
Edit should be restricted to what fields they can edit.

I think the idea was to restrict various fields per permission but I don't really like that idea as it makes things non-standard.  I believe that we should just use the read/edit permissions like everything else where you can read/edit the whole user or not.

As for:
```
permissions.user.editSelf
```

Agreed that is redundant now, we should use an upgrade (I think), we should check when that was added to the codebase as it may just be for 4.0 which would imply we can remove it without an upgrade path.  If we need to do the upgrade then I guess it would be something like this:

1. Get permission roles
2. Find users with those roles
3. Ensure that those users can still edit self by placing those roles onto the edit permission

We should probably think that through further but just tossing out my thoughts here for now.

jazdw, 2021-03-01:
I dont think there is an upgrade path, besides creating a new role for every single user with permission to edit themselves... We can't copy roles from the "edit self" permission to the user's edit permission as this would allow someone to edit another user which they didn't previously have access to edit.

jazdw, 2021-04-01:
Merged into main, primary commits are 31afa163745344c2dc43cff78e5a0c6d12c347d1 and 60e34855f920b2e583bba698916573ca3ce5af82

terrypacker, 2021-05-12:
Released in beta.13

closed 2021-05-12 by terrypacker