MangoAutomation/ma-core-public #1553: Ability for User to send email to request account registration

terrypacker, 2020-03-05:
This will be an endpoint that generates a token with this information to be used when the User is created upon registration:

Organization
Roles

When a publicly registered user registers an email is then sent to the User who generated the link/token/email.  If it is just a public registration then that information is not present in the registration token and no email is sent and the user just remains disabled.

This implies that we will be actioning a new user being created and automate the activation process by sending an email to whoever generated the link to send to the new user.

Joelhaggar, 2020-03-05:
I think read & edit permissions and the creating users email also need to be in the token.

terrypacker, 2020-06-15:
Not doing this, removed from project scope.

closed 2020-06-15 by terrypacker