MangoAutomation/ma-core-public #1563: None level event label is confusing

terrypacker, 2020-03-17:
We have 3 alarms levels that seem to cause confusion.

1.  None - Just a label for the lowest event level 
2.  Do Not Log - Does not go into DB, does not raise user events i.e. web socket messages, does get handled and does go into active events list 
3.  Ignore - Do absolutely nothing

We are proposing renaming 1 to 'Lowest' to indicate that it is definitely a normal event type that does not fall into a special class.

Joelhaggar, 2020-03-18:
I kinds of like "Not Important", I don't think Lowest will really make sense.

terrypacker, 2020-03-18:
@jazdw and @Joelhaggar how about "Low Importance" maybe a little more wordsmithy.

jazdw, 2020-03-18:
I don't like including the word "important" in it. It conveys that there is a relationship with the important level. I think a level named "lowest" makes more sense.

Joelhaggar, 2020-03-19:
What is wrong if someone sees a relationship with important and unimportant?  I think all the alarm levels relate to each other and Not Important or Low Importance would be logical to have if we also have Important, warning, etc.  Lowest doesn't assign the alarm a value and is more similar to "none"  in my mind.  From the user perspective, Not Important will create fewer questions than "Lowest"

jazdw, 2020-03-25:
Your call.

Craig-IAS, 2020-03-25:
I think "Minor" is descriptive and fits in with the rest of the levels

terrypacker, 2020-04-06:
Changed to Minor

closed 2020-04-06 by terrypacker