MangoAutomation/ma-core-public #1564: Script Context Variable Model

terrypacker, 2020-03-17:
In the removal of the v1 REST api there are differences in the script context variable model.  Previously the model used `dataPointXid` to hold the xid of the context point.  In v2 we now use `xid`. I found that you cannot add context points to a meta data point because it is putting the xid into `dataPointXid` which does not end up on the backend.

We need to review the UI for:

- [ ] Meta data source
- [ ] Scripting data source
- [ ] Email event handler
- [ ] Set point event handler

guetteluis, 2020-04-03:
this is happening only in **Meta data source** and it is because of `context-var-xid-name` in here:

```
<ma-script-context 
            name="context"
            context-var-xid-name="'dataPointXid'" 
            updates-context="true" 
            ng-model="$ctrl.dataPoint.pointLocator.context" 
            ng-change="$ctrl.updateContext()"
            ma-get-ctrl="contextModelCtrl = $ngModel">
</ma-script-context>
```

I will create a branch and make PR to fix this

terrypacker, 2020-04-03:
Fixed via https://github.com/infiniteautomation/ma-modules-private/pull/164 

The only problem was with meta points.

pierpuccini, 2020-04-07:
Below showing that scripting validation already includes the xid key and not dataPointXid
[image]

pierpuccini, 2020-04-07:
Below showing that Email event handler validation already includes the xid key and not dataPointXid

[image]

pierpuccini, 2020-04-07:
Below showing that Set point event handler validation already includes the xid key and not dataPointXid

[image]

closed 2020-05-05 by terrypacker