MangoAutomation/ma-core-public #1572: Remove Null Checks on PermissonHolder in Services

terrypacker, 2020-04-30:
This code can be removed as the PermissionService will pick that up now.

```
PermissionHolder user = Common.getUser();
Objects.requireNonNull(user, "Permission holder must be set in security context");
```

jazdw, 2020-07-27:
- User should no longer be null, but an instance of Anonymous
- Don't throw NPE but PermissionException

jazdw, 2020-07-27:
Also ensure we do not check if a user is null anywhere as a part of access control.

terrypacker, 2020-12-03:
Remove the `runAs...` methods from the service layer and assume anytime the SecurityContext permission holder is null that we run with superadmin permission.  See #1624

jazdw, 2021-02-08:
Fixed via e27a3389ede6aaddc17f52956378fc9f116792c2

closed 2021-02-08 by jazdw