MangoAutomation/ma-core-public #1579: Remove getInstance() static methods from DAOs

jazdw, 2020-05-05:
Should be injected via constructor or if that is not possible use `Common.getBean()`

terrypacker, 2020-07-09:
The reason that exists is for the Mango Java Script environment.  Many legacy scripts use code like this:

terrypacker, 2020-07-31:
We have a few options on what to do about the missing 'instance' for DAO access.  Either or both should be documented in the  "Upgrading to Mango 4.0" document.

1.  Put the services into the context (Probably best option)

2.  For those who still want to use DAO access this would be the code:

```
//Example how to get a user
var USERDAO = Java.type("com.serotonin.m2m2.db.dao.UserDao"); 
var userDaoClass = USERDAO.class;
var COMMON = Java.type("com.serotonin.m2m2.Common"); 
var commonClass = COMMON.class;
var userDao = commonClass.static.getBean(userDaoClass);
print(userDao.get(1).name);

//Example how to get the runtime manager state:
var COMMON = Java.type("com.serotonin.m2m2.Common"); 
var commonClass = COMMON.class;
print(commonClass.static.runtimeManager.state);
```

closed 2020-12-04 by terrypacker