MangoAutomation/ma-core-public #1588: H2 - Use MV Store

terrypacker, 2020-07-09:
I found some bugs around missing LOB entries while importing and using large configurations, see https://github.com/infiniteautomation/stallion/issues/40

Using ;MV_STORE=TRUE fixes this problem... I though we upgraded to the MVStore from the PageStore when we upgraded H2

infiniteautomation/ma-core-public#1344

Apparently we did not change to MVStore but I'm not sure why? We just upgraded the database to a higher version that didn't have the LOB expansion bug in the PageStore.

I suggest that we strongly consider moving to the MVStore, I've been using it for a few weeks now and see better performance and no more missing LOB entry messages.

terrypacker, 2020-08-20:
Just a note when choosing mv store vs page store the filename is different.  So there is no concern of db corruption or failure, worst case is Mango starts up on a fresh db.

jazdw, 2020-08-24:
Implemented in 09e501e70c9f5595fc04c2e0da975ab604a5406b

> Just a note when choosing mv store vs page store the filename is different. So there is no concern of db corruption or failure, worst case is Mango starts up on a fresh db.

In fact it seems to use the old page store DB (.h2.db) if it is present even if you pass MV_STORE=TRUE to the driver.

jazdw, 2020-08-24:
I have tested the H2 upgrades for
- Mango 3.5.x (H2 v196)
- Mango 3.7x (H2 v199)

There is some related code in MangoNoSqlService which I still need to look at, I guess this is for migrating point values from H2 to NoSQL.

closed 2020-08-24 by jazdw