MangoAutomation/ma-core-public #1592: RQLUtils - DEFAULT_NODE can get modified

terrypacker, 2020-07-21:
I just found that the RQL Utils class can end up with a corrupted default node which then makes all empty queries use the new restriction.  The only way to recover is to restart Mango.

On line 373 of UserRestController is a good example of how this happens:
```
        if (!service.getPermissionService().hasAdminRole(user)) {
            // Add some conditions to restrict based on user permissions
            rql = RQLUtils.addAndRestriction(rql, new ASTNode("eq", "id", user.getId()));
        }
```

One solution is to not keep a reference to this node and just create an new and node instead of using this.

jazdw, 2020-07-21:
Probably should backport this and release for 3.7

closed 2020-07-21 by jazdw

terrypacker, 2020-07-21:
@jazdw I don't think this can happen in 3.7.  I took a look yesterday and the default node is a limit node so those methods will always create a new node and not add an argument to it.

jazdw, 2020-07-21:
OK cool, that's good then

commit 95f2fe32 (2020-07-21): Fix default RQL node might be modified

Closes #1592