MangoAutomation/ma-core-public #1604: Role REST Behavior

terrypacker, 2020-08-06:
- [ ] expose inherited roles on a user in the model
- [ ] send User web socket notification when a user's roles have changed UPDATE
- [ ] Ensure that the Role websocket sends updates for any roles whose inherited roles have changed

terrypacker, 2020-08-07:
User model change: e2e7328f13735ff5152a8da850930eaca1981710

jazdw, 2020-08-07:
FYI role WebSocket does not work at all. I do not get any create/update/delete notifications on it.

terrypacker, 2020-08-11:
@jazdw I suspect the inherited roles in the model is wrong, we should discuss this.  It is labeled on the permission holder to be `The roles for this permission holder, including all inherited roles` which is likely not exactly what you wanted.

closed 2020-08-13 by terrypacker