MangoAutomation/ma-core-public #1605: Default / built in roles cannot be edited

jazdw, 2020-08-07:
We should be able to update the name of the built in roles, especially since they are not currently localized. We should also discuss if we should allow the user role to inherit other roles too.

terrypacker, 2020-08-13:
* With the change of moving the inherited roles out of the permission holder it would be very possible to allow inheritance.
* The only place the names are hard coded is in the DB create script the references in `PermissionHolder` are to the `Role` not the `RoleVO` which is where the name resides.  So also very possible.

Both of these changes can be done after the Beta release and on the backend would only involve changing the service validation code.  @jazdw how about the frontend?  Shall we just put this off?

terrypacker, 2021-06-30:
Moved to jira https://radixiot.atlassian.net/browse/RAD-1024

closed 2021-06-30 by terrypacker