MangoAutomation/ma-core-public #1617: Require the admin user to change their password on first login

jazdw, 2020-08-17:
We have the ability to require a user to change their password and a password change page. We should require the admin user to do this the first time they login. We will probably have to look at how to make this work with the license agreement page which usually shows on first login.

We could have a bypass for development mode so we don't annoy our developers with this feature.

colbycocking, 2021-05-20:
@jazdw Thought I would remind you about this if mango is supposed to be launching next month. Sorry if I am bringing up stuff you are already thinking about.

jazdw, 2021-05-25:
@colbycocking Yep I am cognizant that there are a lot of things due for Mango 4.0.. Only so many hours in the day :)

@MertCingoz you might be able to take care of this one. I agree with Colby that it is pretty important.