MangoAutomation/ma-core-public #1634: Install Script - URL For latest core

terrypacker, 2020-08-24:
There are 2 new endpoints for the store at:
 
`/free-m2m2-core-latest.zip`
`/enterprise-m2m2-core-latest.zip`

We can change the behavior but currently this will return the latest version including non-production cores.  The logic is contained in the store class `LatestBundleDownloadServlet`

jazdw, 2020-09-01:
@terrypacker did we add a URL that returns the latest release version number?

terrypacker, 2020-09-01:
@jazdw I did not but I noticed these were committed some time ago and is now live on the store after my last update:

```
https://store.infiniteautomation.com/enterprise-m2m2-core-latest.zip
https://store.infiniteautomation.com/free-m2m2-core-latest.zip
```

jazdw, 2020-09-01:
OK well the script is going to be a bit crappy still in that it will only give you the option to download the "latest" release. It will not be able to tell you "latest (4.0.0)" for example.

terrypacker, 2020-09-01:
I'm not sure if we want to use them or not.  The "latest" bundles can now be pre release versions so currently that will get you the 4.0.0-beta.1.

It won't be hard to add an endpoint, which I can do.  I suspect we should incorporate some sort of classifier on the request to determine what channel you want to use?

jazdw, 2020-09-01:
> I'm not sure if we want to use them or not. The "latest" bundles can now be pre release versions so currently that will get you the 4.0.0-beta.1.

Do not want. I just want a link for the latest actual released version.

> It won't be hard to add an endpoint, which I can do. I suspect we should incorporate some sort of classifier on the request to determine what channel you want to use?

No this is not needed either. If they want a particular version they can still enter a string.

terrypacker, 2020-09-01:
Can you be specific on exactly what you want from the endpoint in the situation there is a non production release available?  So currently you would either want:

`4.0.0-beta.1` or `4.0.0`

jazdw, 2020-09-01:
Also URLs should be under downloads/fullCores as per https://store.infiniteautomation.com/downloads/fullCores/${MA_BUNDLE_TYPE}-m2m2-core-${MA_VERSION}.zip"

jazdw, 2020-09-01:
> Can you be specific on exactly what you want from the endpoint in the situation there is a non production release available? So currently you would either want:

I want the current released Mango version. Right now that would be `3.7.7`

terrypacker, 2020-12-03:
See infiniteautomation/m2m2Store#33