MangoAutomation/ma-core-public #1643: Error when inserting user comment via UI

terrypacker, 2020-09-25:
```
ERROR 2020-09-25T21:05:57,276 (com.infiniteautomation.mango.rest.latest.exception.RestExceptionHandler.handleExceptionInternal:248) - REST request status 500 INTERNAL_SERVER_ERROR:
org.springframework.dao.DataIntegrityViolationException: PreparedStatementCallback; SQL [insert into userComments (`xid`, `userId`, `ts`, `commentText`, `commentType`, `typeKey`) values (?, ?, ?, ?, ?, ?)]; Column 'commentText' cannot be null; nested exception is java.sql.SQLIntegrityConstraintViolationException: Column 'commentText' cannot be null
	at org.springframework.jdbc.support.SQLExceptionSubclassTranslator.doTranslate(SQLExceptionSubclassTranslator.java:87) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.jdbc.support.AbstractFallbackSQLExceptionTranslator.translate(AbstractFallbackSQLExceptionTranslator.java:72) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.jdbc.support.AbstractFallbackSQLExceptionTranslator.translate(AbstractFallbackSQLExceptionTranslator.java:81) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.jdbc.core.JdbcTemplate.translateException(JdbcTemplate.java:1443) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:633) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at com.serotonin.db.spring.ExtendedJdbcTemplate.update(ExtendedJdbcTemplate.java:113) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.db.spring.ExtendedJdbcTemplate.update(ExtendedJdbcTemplate.java:96) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.db.spring.ExtendedJdbcTemplate.update(ExtendedJdbcTemplate.java:88) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.db.spring.ExtendedJdbcTemplate.doInsert(ExtendedJdbcTemplate.java:157) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.db.dao.AbstractBasicDao.lambda$insert$1(AbstractBasicDao.java:202) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at org.springframework.transaction.support.TransactionTemplate.execute(TransactionTemplate.java:140) ~[spring-tx-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at com.serotonin.db.DaoUtils.doInTransaction(DaoUtils.java:306) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.db.dao.AbstractBasicDao.insert(AbstractBasicDao.java:196) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.db.dao.AbstractVoDao.insert(AbstractVoDao.java:157) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.m2m2.db.dao.AbstractVoDao.insert(AbstractVoDao.java:28) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.infiniteautomation.mango.spring.service.AbstractVOService.insert(AbstractVOService.java:105) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.infiniteautomation.mango.rest.latest.UserCommentRestController.createNewUserComment(UserCommentRestController.java:125) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:?]
	at jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:?]
	at jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:?]
	at java.lang.reflect.Method.invoke(Method.java:566) ~[?:?]
	at org.springframework.web.method.support.InvocableHandlerMethod.doInvoke(InvocableHandlerMethod.java:190) ~[spring-web-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.method.support.InvocableHandlerMethod.invokeForRequest(InvocableHandlerMethod.java:138) ~[spring-web-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.ServletInvocableHandlerMethod.invokeAndHandle(ServletInvocableHandlerMethod.java:105) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.invokeHandlerMethod(RequestMappingHandlerAdapter.java:879) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.annotation.RequestMappingHandlerAdapter.handleInternal(RequestMappingHandlerAdapter.java:793) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.mvc.method.AbstractHandlerMethodAdapter.handle(AbstractHandlerMethodAdapter.java:87) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:1040) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:943) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:1006) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.servlet.FrameworkServlet.doPost(FrameworkServlet.java:909) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:883) ~[spring-webmvc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) ~[javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:755) ~[jetty-servlet-9.4.27.v20200227.jar:9.4.27.v20200227]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1617) ~[jetty-servlet-9.4.27.v20200227.jar:9.4.27.v20200227]
	at org.springframework.web.filter.CharacterEncodingFilter.doFilterInternal(CharacterEncodingFilter.java:201) ~[spring-web-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:119) ~[spring-web-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1604) ~[jetty-servlet-9.4.27.v20200227.jar:9.4.27.v20200227]
	at com.infiniteautomation.mango.webapp.filters.MangoCacheControlHeaderFilter.doFilter(MangoCacheControlHeaderFilter.java:155) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1604) ~[jetty-servlet-9.4.27.v20200227.jar:9.4.27.v20200227]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:320) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.authentication.switchuser.SwitchUserFilter.doFilter(SwitchUserFilter.java:200) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:126) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:90) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at com.serotonin.m2m2.web.mvc.spring.security.RateLimitingFilter.doFilterInternal(RateLimitingFilter.java:91) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:119) ~[spring-web-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at com.serotonin.m2m2.web.mvc.spring.security.PermissionExceptionFilter.doFilter(PermissionExceptionFilter.java:34) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:118) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:334) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:137) ~[spring-security-web-5.3.4.RELEASE.jar:5.3.4.RELEASE]
[... cut ...]
	at org.apache.commons.dbcp2.DelegatingPreparedStatement.executeUpdate(DelegatingPreparedStatement.java:136) ~[commons-dbcp2-2.7.0.jar:2.7.0]
	at com.serotonin.db.spring.ExtendedJdbcTemplate$1.doInPreparedStatement(ExtendedJdbcTemplate.java:121) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at com.serotonin.db.spring.ExtendedJdbcTemplate$1.doInPreparedStatement(ExtendedJdbcTemplate.java:113) ~[mango-4.0.0-SNAPSHOT.jar:?]
	at org.springframework.jdbc.core.JdbcTemplate.execute(JdbcTemplate.java:617) ~[spring-jdbc-5.2.6.RELEASE.jar:5.2.6.RELEASE]
	... 115 more
```

terrypacker, 2020-09-25:
This should be validated before insertion.

Pacane, 2020-11-09:
@terrypacker the server side portion of this is done. Should we keep it open for someone to do the frontend part?

terrypacker, 2020-11-09:
@Pacane did you test what happens when you enter empty text?  The UI should highlight the text input with a message that says required.  If that does not happen then we should open an issue in `ma-dashboards` for this.

terrypacker, 2020-12-03:
Fixed with 23395033daeadab09f54f32ff59c0558af0562ce

closed 2020-12-03 by terrypacker