MangoAutomation/ma-core-public #1654: Point value WebSocket updates not sent for scripting data source

jazdw, 2020-10-22:
When setting a point value via the scripting data source like `point.set(true)`, a point value WebSocket update event is not sent.

terrypacker, 2020-10-28:
@jazdw I just tested this and found that when you have the 'Historical' checkbox checked you won't get events.  With this checked no events will be fired, thus no websocket message will be emitted.  However I suspect that you have that set so you can set values on external points, which in the code does this, for reasons unknown to me:

```
//If we're not going through the data source, we don't care if it's settable, really.
if(!vo.getHistoricalSetting() && !point.getVO().getPointLocator().isSettable() && dprt.getDataSourceId() != vo.getId())
    return;
```

This firing of events behavior is debatable, but was done in the situation where one would be generating "tons" of historical values and websocket messages would overload the page.  

This could be changed to expose a new method on the point that lets you fire events when setting, something like:

```
var fireEvents = ALWAYS; //[ALWAYS, ON_CURRENT_VALUE_UPDATE, NEVER]
point.set(true, ALWAYS);
```

terrypacker, 2020-10-28:
Also adding a task here to add help about this "feature"
 - [ ] add notes about Historical setting to help file

jazdw, 2020-10-28:
OK the "Saves historic" checkbox was checked. That explains it enough for me. Changed the translation so the checkbox matches the documentation.

closed 2020-10-28 by jazdw