MangoAutomation/ma-core-public #1662: Move files from classes to resources

jazdw, 2020-11-06:
So that i18n.properties, env.properties etc are loaded from jar file rather from disk. We should then implement some way to generate an empty config file.

jazdw, 2020-12-04:
Leaving this open until we create a way to generate a env.properties on first start. This would look for the overrides/properties/env.properties file or the file specified by MA_ENV_PROPERTIES and create it if it doesn't exist (commenting out all lines).

jazdw, 2020-12-04:
Module i18n.properties are still located in the classes directory but are no longer copied to zip file. We should move these files into resources.

terrypacker, 2020-12-04:
We should change any documentation about copying this file from the classes folder to say that this file will be automatically copied across to the overrides.

- [ ] make this file r/w for the mango user

terrypacker, 2020-12-04:
Note also that the log4j2.xml file has been moved to the jar, update documentation as necessary.

terrypacker, 2020-12-04:
- [ ] Decide what to do about existing `classes` directory on upgrade.  It won't be on the class path

terrypacker, 2021-05-13:
Released and in production

closed 2021-05-13 by terrypacker