MangoAutomation/ma-core-public #1676: Mango Startup Query for Groups of Data Points is wrong

terrypacker, 2020-12-13:
The query looks something like this:

```
SELECT id, value,timestamp FROM pointValues WHERE id = ANY ($1) AND timestamp<$2 ORDER BY
timestamp DESC LIMIT 1
```

This limit of 1 is concerning.  Can we possibly limit to 1 per id?

jazdw, 2020-12-14:
Obviously this is something that needs to be fixed, however I doubt this is impacting anyone right now. We should be using iasTsdb or Timescale yeah?

jazdw, 2020-12-16:
b38b9f01155fa933d99f120a6ae5e3c0ff87ff33 orderedById is required

jazdw, 2020-12-17:
Implemented for PointValueDaoSQL in 582a83667234a1084f05b077960930eed20ec6c1 and then 6ad12fc46ac11fb4e55585dec0d2576d7991af16

Timescale still needs to be implemented.

terrypacker, 2020-12-17:
See also for TimeScale https://github.com/infiniteautomation/ma-modules-proprietary/commit/f9126ba7b154f37cdee5e963b5b9f580d93bc413

terrypacker, 2021-01-07:
Fixed

closed 2021-01-07 by terrypacker