MangoAutomation/ma-core-public #1677: Mango Data Point Group Initializer Multi threading problem

terrypacker, 2020-12-13:
These were noticed on a very large system using groups of 1000 and 32 threads:

1.  The CPU usage was at 100%, I would have expected something near 3200%
2. The processor was hanging if one of the group queries was slow.  I suspect it performs execution on groups of 32 at a time and waits for all 32 to finish before submitting a new group of 32.

See #1676 as that needs to be solved first.

jazdw, 2020-12-14:
> The CPU usage was at 100%, I would have expected something near 3200%

Depends on the tool you are using.

> I suspect it performs execution on groups of 32 at a time and waits for all 32 to finish before submitting a new group of 32.

It does not, well at least that is not the way it was designed. As soon as one thread finishes it should pick up the new task. I am pretty confident that this works correctly, you can see this behavior looking at the start up logs.

terrypacker, 2020-12-14:
Not really what is happening, this is very likely all 32 threads stuck on long running queries.

closed 2020-12-14 by terrypacker