MangoAutomation/ma-core-public #1678: User Session Expiration Setting - Not being used correctly in Jetty

terrypacker, 2020-12-14:
The system setting for the session expiration sets the `max-age` and `expiration` for the cookie to the correct value in the `MangoSessionHandler` class.  However the per-user expiration is setting the wrong thing and not anything on the cookie.

The per user setting is applied wrongly in the `MangoAuthenticationSuccessHandler` like so:

```
        //Set the per-user session timeout
        if(user.isSessionExpirationOverride()) {
            HttpSession session = request.getSession(false);
            if (session != null) {
                session.setMaxInactiveInterval((int)(Common.getMillis(Common.TIME_PERIOD_CODES.getId(user.getSessionExpirationPeriodType()), user.getSessionExpirationPeriods())/1000L));
            }
        }
```

Which is not the session timeout but the inactive interval.

jazdw, 2020-12-21:
Further to this, the session cookie `Max-Age` is set to whatever the system setting "Session timeout" is set to but is never updated. This means that even if the user is actively using the session the session cookie is removed by the browser and the user will have to log in again. I can see that with session persistence turned on the actual session on the backend is still valid and the `expiryTime` column is being update on every request.

jazdw, 2020-12-21:
- [ ] decrease default session timeout?
- [ ] Update active session's maxInterval when the session timeout is changed via system settings or via user page

jazdw, 2021-01-05:
Tracking remaining task in #1684

closed 2021-01-05 by jazdw